0xFireWolf/RealtekCardReader #1: Immediate KP when injecting or caching this beta kext in Big Sur 11.4

RV-ABZ, 2021-06-19:
Whether using the Debit or Release version 0.9.0, IO experience a systematic immediate KP on booting Big Sur 11.4. OS hardly begins to load so there's no log available afterwards.

Test platform is a Dell Latitude E7270 laptop (Skylake) fitted with Realtek RTS525A SD card reader:
Vendor id: 10ec
Device id: 525a
Subsys Ven id: 1028
Subsys id: 06db
Revision: 1

[image]
[image]
[image]

0xFireWolf, 2021-06-19:
v0.9.1 should fix the issue. Please give a try and see whether the issue still exists.

0xFireWolf, 2021-06-21:
Resolved.

closed 2021-06-21 by 0xFireWolf