0xFireWolf/RealtekCardReader #2: About RTS5227.

CLAY-BIOS, 2021-06-21:
Version: RealtekPCIeCardReader.kext 0.9.2
My card reader is RTS5227, but it is not in the supported chip.
I noticed that the two chips RTS5227 and RTS5249 are very similar.
I added RTS5227 to info.plist and used class RealtekRTS5249Controller to make RTS5227 work.
[image]
[image]
1. RTS5227 can work normally in macOS12, but it will cause kp in macOS10.15.
[10.15-kp.md.zip](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6685298/10.15-kp.md.zip)
2. Since there is no power management, the card reader will not work after sleep wakeup.

0xFireWolf, 2021-06-21:
Note that each controller class contains chip-specific initialization routine. RTS5227 is similar but not identical to RTS5249, so you should not use RealtekRTS5249Controller for your card reader.

RTS5227 and RTS522A will be supported as of 0.9.3. I already pushed the code to the repo, so you can compile the driver by yourself and try the new one.

CLAY-BIOS, 2021-06-21:
> Note that each controller class contains chip-specific initialization routines. RTS5227 is similar but not identical to RTS5249, so you should not use RealtekRTS5249Controller for your card reader.
> 
> RTS5227 and RTS522A will be supported as of 0.9.3. I already pushed the code to the repo, so you can compile the driver by yourself and try the new one.
I compiled the latest version, but the card reader did not work.
[image]
[image]
There will be a large amount of RTSX starting information when booting.
[image]
[image]

0xFireWolf, 2021-06-21:
> I compiled the latest version, but the card reader did not work.

Your IOReg indicated that the chip was initialized successfully. 

If your SD card could not be recognized, please refer to the FAQs and attach the kernel log.

CLAY-BIOS, 2021-06-21:
> > I compiled the latest version, but the card reader did not work.
> 
> Your IOReg indicated that the chip was initialized successfully.
> 
> If your SD card could not be recognized, please refer to the FAQs and attach the kernel log.
> 
> Also, please insert your card after the system boots. It is a known issue that the card may not be initialized while the system is booting.

The SD card cannot be identified，This is my log.
[rtsx.log](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6688229/rtsx.log)

0xFireWolf, 2021-06-24:
> The SD card cannot be identified，This is my log.
> [rtsx.log](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6688229/rtsx.log)

The host driver failed to switch to the 4-bit bus because your card does not respond to ACMD6.  
I updated the code to retry up to 10 times if the response to an ACMD is timed out.   
Please try v0.9.3b2 and see if it solves the problem. 

[RealtekPCIeCardReader_0.9.3v2_DEBUG.zip](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6707832/RealtekPCIeCardReader_0.9.3v2_DEBUG.zip)

Also, your kernel log is incomplete.   
Please Refer to the FAQs to make sure that it contains all necessary information.

CLAY-BIOS, 2021-06-24:
> The host driver failed to switch to the 4-bit bus because your card does not respond to ACMD6.
> I updated the code to retry up to 10 times if the response to an ACMD is timed out.
> Please try v0.9.3b2 [...]

The sd card is still not recognized, I added DebugEnhancer.
This is my log.
[rtsx.log](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6708300/rtsx.log)

seungjunProgramming, 2021-06-24:
Will upload the log soon

0xFireWolf, 2021-06-24:
> The sd card is still not recognized, I added DebugEnhancer.
> This is my log.
> [rtsx.log](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6708300/rtsx.log)

> Here's the log(kext loaded)
> [rtsx.log](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6710335/rtsx.log)

Neither of your kernel logs is complete. Am I supposed to guess what's wrong with a chip that I don't own?

Check the FAQs and see what a complete kernel log looks like at the beginning.  
Please make sure that your logs are indeed complete before you upload it next time.

seungjunProgramming, 2021-06-26:
Hm... I'm sorry but I have installed the debug enhancer and have put in the msgbuf=10485760 at nvram.
But, This gives me an empty log...
I'm sorry but what should I do???

seungjunProgramming, 2021-06-26:
[rtsx.log](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6720375/rtsx.log)
Here is my log

seungjunProgramming, 2021-06-27:
Looks like I/O timeout at IOReturn(kIOReturnTimeout)

0xFireWolf, 2021-07-02:
> [rtsx.log](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6720375/rtsx.log)
> 
> Here is my log
> 
> I built my own kext with target OS 12.0 and used it

Your log is still incomplete.

Anyway, take a look at [0.9.3v4](https://www.insanelymac.com/forum/topic/348130-firewolf-realtek-pcie-card-reader-driver-for-macos/?do=findComment&comment=2762133) and see whether it fixes your issue as well. Or you can compile the latest version by yourself. If it does not work, please attach the **COMPLETE** kernel log.

seungjunProgramming, 2021-07-05:
@0xFireWolf Um... This is strange
I get firmware loaded at the boot log when I boot, but I get a empty log when I extract the log.
(when I add the debugenhancer.kext and msgbuff at NVRAM, I get an empty log) 
(when I boot without those two, I get an incomplete log)

seungjunProgramming, 2021-07-05:
Also, When I tested it, it successfully loads the card present boolean in ioRegisteryexplorer.
(When I remove the card: False When I insert the card: True), but card was not readable

seungjunProgramming, 2021-07-10:
Here's my full log
[rtsx.log](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6794597/rtsx.log)

seungjunProgramming, 2021-07-10:
Oops... Uploaded the wrong log
Here's my COMPLETE LOG(I guess)
@0xFireWolf 
[rtsx.log](https://github.com/0xFireWolf/RealtekPCIeCardReader/files/6794616/rtsx.log)

0xFireWolf, 2021-08-20:
Hello @CLAY-BIOS, if you are still interested in testing this kext, please check whether the latest version fixes your timeout issue after the driver switches the card clock. You can find compiled version under [`Artifacts`](https://github.com/0xFireWolf/RealtekCardReader/actions/runs/1151885389).

0xFireWolf, 2021-08-28:
> Hello @CLAY-BIOS, if you are still interested in testing this kext, please check whether the latest version fixes your timeout issue after the driver switches the card clock. You can find compiled version under [`Artifacts`](https://github.com/0xFireWolf/RealtekCardReader/actions/runs/1151885389).

Based on the kernel logs uploaded by other beta testers, the timeout issue after the driver switches the card clock is resolved by the latest 0.9.6 release. Issue closed. If the original poster still has the same issue with RTS5227, please create a new issue with a **complete** kernel log.

closed 2021-08-28 by 0xFireWolf