0xFireWolf/RealtekCardReader #3: Adding RTS5260 Support

lulujyc, 2021-06-29:
Hi FireWolf:

First of all, thank you very much for implementing the Realtek SD card driver. I'm testing it on a Dell Precision 7530, which has the RTS5260 chip (10EC, 5260). I used the RTS525A class to experiment, however, it is not supported and recognized at all. I wonder if there is something I could do to help you implement the driver. The chip is most commonly found on Dell business laptops. Thank you again!

0xFireWolf, 2021-07-16:
> I used the RTS525A class to experiment, however, it is not supported and recognized at all.

This is expected, because each controller class contains chip-specific code that may not be compatible for other chips.

Try the attached driver and see if your chip can be recognized and work properly.

[RealtekCardReader_0.9.6p1_f2b10c6_DEBUG.zip](https://github.com/0xFireWolf/RealtekCardReader/files/6829339/RealtekCardReader_0.9.6p1_f2b10c6_DEBUG.zip)

lulujyc, 2021-07-17:
Thank you so much! The card reader worked well on Monterey, with your build and RealtekCardReaderFriend.

[image]

closed 2021-07-17 by lulujyc

0xFireWolf, 2021-07-17:
Does the SD card work properly as well? What's the read and write speed under macOS and other systems like Windows?

lulujyc, 2021-07-17:
Yes, it is correctly recognized and has moderate writing speeds, as I only have an SDHC card to test. Sorry I don't have any UHS cards to report the maximum speed of the driver.