0xFireWolf/RealtekCardReader #4: RTS522A unable to initialize card

seungjunProgramming, 2021-07-18:
When I boot, the kext is correctly attached to the reader, but when I insert the card, it fails to initialize the card even with the default frequency, and fails to read the card.
Here's my log.
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/6836324/rtsx.log)
(This is the log made with the 1bit code that you uploaded at the insainlymac today. I'll present the log with the uploaded 0.9.5 kext if needed!)

seungjunProgramming, 2021-07-19:
Here's the log made with the latest commit(5842de3). I built it with MacKernelSDK inside the folder and changed the release OS to 12.0 (Using Xcode beta)

seungjunProgramming, 2021-07-19:
Tell me if I have any missing or incomplete logs please!

0xFireWolf, 2021-07-23:
Your kernel log indicated that the card became unresponsive (CMD2: timeout) after it switched to 1.8V (CMD11: success).

Unfortunately, I don't have a clue at this moment, but it might be worth trying to initialize the card at 3.3V instead.

seungjunProgramming, 2021-07-25:
I recently tried another kext madly another developer, and it successfully read my card. 
Will I be able to read the card if I find out what V it uses to initialize my card?
I'm so sorry about mentioning another developer's kext, but I thought it may be a hint.
This is the kext I used.
https://github.com/cholonam/Sinetek-rtsx

0xFireWolf, 2021-07-25:
> Will I be able to read the card if I find out what V it uses to initialize my card?

If I remember correctly, your card is probably operating at 50 MHz under 3.3V with that driver. 

Do you have any other SDHC or SDXC cards to test with my driver? Will those cards become unresponsive as well after switching to 1.8V?

> I'm so sorry about mentioning another developer's kext, but I thought it may be a hint.

No need to apologize. I am aware of that driver, and Cholonam worked very hard to improve Sinetek's original driver to support RTS525A. Unfortunately, it supports the default and the high speed modes only, so voltage switches are not needed.

seungjunProgramming, 2021-07-25:
I have a SDHC I 16GB UHS-1 Card.
I'll upload the log right away

seungjunProgramming, 2021-07-25:
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/6873852/rtsx.log)
Here's my log.
It was also undetected...

0xFireWolf, 2021-07-26:
[RealtekCardReader_0.9.6_4213fa4_DEBUG.zip](https://github.com/0xFireWolf/RealtekCardReader/files/6874896/RealtekCardReader_0.9.6_4213fa4_DEBUG.zip)

Add the boot argument `-rtsx3v3` to initialize the card at 3.3V and attach the kernel log.

seungjunProgramming, 2021-07-27:
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/6883723/rtsx.log)
Here's my log.
It successfully checks the card(GB, etc...), but can not read the card.
[image]
[image]

seungjunProgramming, 2021-07-27:
[image]

0xFireWolf, 2021-08-20:
Hello there. Your card also timed out after the driver switched the clock to 50MHz. Please check whether the latest version fixes your issue. You can find compiled version under [`Artifacts`](https://github.com/0xFireWolf/RealtekCardReader/actions/runs/1151885389).

lekoOwO, 2021-08-25:
Same issue here. Any guides to provide logs?

lekoOwO, 2021-08-25:
Same issue here. Here's the log using nightly build.

[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/7045783/rtsx.log)