0xFireWolf/RealtekCardReader #7: 525a driver problem

jxweb, 2021-07-26:
Hello:
At present, there is a problem with the 525a, that is, if the SD card is inserted into the computer before booting, it cannot be driven. I don't know if this is the case for me alone?

0xFireWolf, 2021-07-26:
Please follow the FAQs and attach your kernel log.
Bug reports without any details and logs will be closed and left unanswered.

jxweb, 2021-07-27:
Sorry, this is the log file。
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/6881926/rtsx.log)

0xFireWolf, 2021-07-27:
Your kernel log is incomplete. 

Please carefully read the FAQs and ensure that the log is complete before uploading it.

jxweb, 2021-07-27:
Hello, added the boot argument msgbuf=10485760, this is related log
thanks！
[rtsx1.log](https://github.com/0xFireWolf/RealtekCardReader/files/6883729/rtsx1.log)
[image]
[image]

jxweb, 2021-07-29:
The previous log is incomplete. Now I have added a complete log. Could you please see what's the problem?
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/6898532/rtsx.log)
[    2.545096]: RTSX: virtual bool RealtekPCICardReaderController::start(IOService *) DInfo: ===================================================
[    2.545100]: RTSX: virtual bool RealtekPCICardReaderController::start(IOService *) DInfo: Starting the Realtek PCIe card reader controller...
[    2.545102]: RTSX: virtual bool RealtekPCICardReaderController::start(IOService *) DInfo: ===================================================
[    2.545121]: RTSX: virtual bool RealtekCardReaderController::start(IOService *) DInfo: =======================================================
[    2.545123]: RTSX: virtual bool RealtekCardReaderController::start(IOService *) DInfo: Starting the Realtek PCIe/USB card reader controller...
[    2.545125]: RTSX: virtual bool RealtekCardReaderController::start(IOService *) DInfo: =======================================================

0xFireWolf, 2021-08-07:
Your card does not respond to the initial ACMD41 while the system is booting, but I was not able to reproduce your issue with my cards. 

I added a boot argument `rtsxdcib=100` to delay a certain amount of time in milliseconds before the driver initializes the card found at boot time. Please try the attached driver and adjust the value (100) if necessary. 

If it still does not work, please insert the card after the system boots. I do not plan to investigate this issue further. Thank you for testing the driver.

[RealtekCardReader_0.9.6_851ede2_DEBUG.zip](https://github.com/0xFireWolf/RealtekCardReader/files/6948579/RealtekCardReader_0.9.6_851ede2_DEBUG.zip)

closed 2021-08-07 by 0xFireWolf