0xFireWolf/RealtekCardReader #8: RTS5129 (0x129 0xbda) issue

gnpaone, 2021-08-03:
Hi @0xFireWolf you are doing a great job in developing the kext for card reader. As you mentioned in 0.9.5 changelog, I did add both the kexts but nothing changed for RTS5129, it's same as before. I am using opencore and I have attached some pics too. You also mentioned that you don't have this card reader with you right? I can help you with whatever tests you need to do with the card reader as I have it but please make it work.
[image]
[image]
[image]
[image]

m0d16l14n1, 2021-08-03:
You can find all info here, there is the topic with testers and test builds. Work in progress, so it's still unknown when it's will work "fully" 

https://www.insanelymac.com/forum/topic/348130-realtek-pcie-card-reader-driver-for-macos/page/4/

gnpaone, 2021-08-03:
Update: I removed the both realtekcardreader and realtekcardreaderfriend kexts and replaced it with genericcardreaderfriend kext by @0xFireWolf and it does show up in card reader list but doesn't detect any scared when inserted (I think it's mentioned generic kext creates a fake sdcard reader probably 🤔) but it's still the same look in "USB" of "system report"

[image]

gnpaone, 2021-08-03:
> @gnpaone You can find all info here, there is the topic with testers and test builds. Work in progress, so it's still unknown when it's will work "fully"
> 
> https://www.insanelymac.com/forum/topic/348130-realtek-pcie-card-reader-driver-for-macos/page/4/

Thanks, I will follow the forum

gnpaone, 2021-08-04:
> > @gnpaone You can find all info here, there is the topic with testers and test builds. Work in progress, so it's still unknown when it's will work "fully"
> > https://www.insanelymac.com/forum/topic/348130-realtek-pcie-card-reader-driver-for-macos/page/4/
> 
> Thanks, I will follow the forum

https://www.insanelymac.com/forum/topic/348130-realtek-pcie-card-reader-driver-for-macos/?do=findComment&comment=2764183

gnpaone, 2021-08-09:
> https://www.insanelymac.com/forum/topic/348130-realtek-pcie-card-reader-driver-for-macos/?do=findComment&comment=2764183 [...]

https://www.insanelymac.com/forum/topic/348130-realtek-pcie-card-reader-driver-for-macos/?do=findComment&comment=2764430  @0xFireWolf

gnpaone, 2021-08-10:
> https://www.insanelymac.com/forum/topic/348130-realtek-pcie-card-reader-driver-for-macos/?do=findComment&comment=2764430 [...]

https://www.insanelymac.com/forum/topic/348130-realtek-pcie-card-reader-driver-for-macos/?do=findComment&comment=2764505 @0xFireWolf

m0d16l14n1, 2021-08-10:
@gnpaone please, stop spamming your own comments. Once dev will have time, he will look, no need to quote and ping every time you post something ^_^

gnpaone, 2021-08-10:
Fine

0xFireWolf, 2021-08-28:
The latest 0.9.6 release has resolved the issue that cards cannot be initialized properly on USB-based card readers.
Issue closed. If the original poster has found other issues with the latest release, please create a new issue with complete kernel logs.

closed 2021-08-28 by 0xFireWolf