0xFireWolf/RealtekCardReader #12: RTS5449 (0x177 0x129) Timeout issue

Ab2774, 2021-09-15:
Hi, this issue is pretty similar to https://github.com/0xFireWolf/RealtekCardReader/issues/8, using GenericCardReaderFriend, System Information recognize the card only when it's plugged in, however, with RealtekCardReader and RealtekCardReaderFriend when I plug my SD card, it times out similar to https://github.com/0xFireWolf/RealtekCardReader/issues/8, I tried several times to get a log with msgbuf=10485760 and the debug version of the kext, but RealtekCardReader doesn't seem to load on my laptop for some reason, only RealtekCardReader loads, I also tried rtsxdssc=N boot-arg but the results are the same, so instead, I uploaded my EFI and IOReg.
Cheers
[Abdulrahim’s MacBook Air.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7172045/Abdulrahim.s.MacBook.Air.zip)
[EFI.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7172048/EFI.zip)

0xFireWolf, 2021-09-15:
Your USB card reader (0x0BDA0177) is not supported by `RealtekCardReader`.

>  using GenericCardReaderFriend, System Information recognize the card only when it's plugged in

This is expected when you are using GenericCardReaderFriend. See the [FAQs](https://github.com/0xFireWolf/GenericCardReaderFriend/blob/main/FAQ.md#long-story).

closed 2021-09-15 by 0xFireWolf

Ab2774, 2021-09-15:
I understand, is there's a chance in the future that it will get supported or should I just start using GenericCardReaderFriend again?

0xFireWolf, 2021-09-15:
GenericCardReaderFriend is for decoration purposes only. 

Isn't your card reader natively supported by Apple's generic mass storage driver?

Ab2774, 2021-09-15:
> GenericCardReaderFriend is for decoration purposes only.
> 
> Isn't your card reader natively supported by Apple's generic mass storage driver?

It's recognized by the system as a normal 2.0USB hub, the card shows as a USB drive, it's great that's it's working with no additional kexts but it would be awesome to get it recognized as an SD Card reader hub

Ab2774, 2021-09-15:
Hi, I know this is not the repo but I'm still experiencing timeout issues with GenericCardReaderFriend, any suggestions?

0xFireWolf, 2021-09-15:
What kind of timeout issues? 
If I remember it correctly, `GenericCardReaderFriend` does not contain any functions that return a timeout error. 

Note that you are **NOT** supposed to use `RealtekCardReader` and `RealtekCardReaderFriend` for your RTS5449. 
Make sure that these two kexts are not installed or injected by your bootloader. 
If the problem still exists, please attach your complete kernel log.

Ab2774, 2021-09-15:
Me too, I don't remember experiencing this issue with GenericCardReaderFriend, I only did with RealtekCardReader and RealtekCardReaderFriend.
[Abdulrahim’s MacBook Air.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7172690/Abdulrahim.s.MacBook.Air.zip)
[EFI.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7172692/EFI.zip)
[kernel.log.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7172694/kernel.log.zip)

0xFireWolf, 2021-09-15:
So, where is the timeout? 
The kernel log you uploaded contains only one line, which is totally useless. 
Your IOReg shows that you inserted the card and then removed it and did that again.
I have no idea what issue you are reporting here.

Ab2774, 2021-09-15:
I'm sorry for the inconvenience, what I'm trying to say is that when I plug my SD card adapter, it shows in System Information for a few seconds and then disappears till I replug it, I don't know if that's expected behavior or not.