0xFireWolf/RealtekCardReader #13: Integrate recognition as a built-in device

antoniomcr96, 2021-09-17:
First of all, thanks for all your work.
The idea that I want to propose consists of merging this kext with _RealtekCardReaderFriend_ in order to always show the card reader as a built in device.
Maybe we can use a boot-arg or (better) inject a deviceproperty to control this feature.
What do you think?
Thanks in advance

0xFireWolf, 2021-09-17:
No, I deliberately choose not to combine them into a single kext, otherwise `RealtekCardReader` always depends on `Lilu` regardless of whether users want to turn on the features provided by `RealtekCardReaderFriend`.

closed 2021-09-17 by 0xFireWolf

antoniomcr96, 2021-09-17:
Thanks for the explaination. Anyway, in a real-life scenario, I think that everyone who needs this kext uses Lilu. Have a nice day :)