0xFireWolf/RealtekCardReader #14: SC Card not showing, kext loaded, card reader shows up in System Report

Narcoroni, 2021-09-17:
Hi,
my Razer Blade Hackintosh card reader isn't working yet.
It's showing in the system report, and in ioReg, but the SD card itself is not shown by MacOS.
Hope you can help me out. 

Opencore 0.7.3
MacOS BigSur 11.5.2
Realtek RTS5260 PCIe card reader

[image]
[image]
[image]
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/7188513/rtsx.log)
.

0xFireWolf, 2021-09-18:
Your kernel log is incomplete. Next time please attach the full log as a file.

However, your kernel log shows that the driver fails to tune the UHS-I SDHC/XC card. 
This is a known issue on RTS5260. Add the boot argument `-iosd3v3` to initialize all cards at 3.3V for now.
See the discussion on [InsanelyMac](https://www.insanelymac.com/forum/topic/348130-realtek-pcieusb-sd-card-reader-driver-for-macos/page/7/?tab=comments#comment-2766019).

Narcoroni, 2021-09-18:
That works.
Thank your for your digital wizardry!

0xFireWolf, 2021-09-18:
You are welcome. Thanks for your feedback.
I will reply to this thread here when I have time to investigate the issue and fix it.
Issue closed for now.

closed 2021-09-18 by 0xFireWolf