0xFireWolf/RealtekCardReader #15: can not shutdown when using RealtekCardReader.kext

mylife1234-cmd, 2021-09-26:
I'm using using dell vostro 3578 laptop 
CPU :I5 8250U
RAM:8GB
INTEL HD 620 
When  I using RealtekCardReader.kext and RealtekCardReaderFriend.kext,  I shutdown my laptop bit it restart
If I remove this kext, It's normal
Can you help me ?

gnpaone, 2021-09-26:
Can u please mention which card reader you have (and upload kernel logs)?

mylife1234-cmd, 2021-09-26:
I don't know. Can you tell me a tool to check my model card reader
I'm newbie 
Thanks you so much

gnpaone, 2021-09-26:
Please go to system information and select "card reader" and send screenshot. Realtekcard reader kexts must be present for this

mylife1234-cmd, 2021-09-26:
[image]
Thanks you very much
This is my screenshot

gnpaone, 2021-09-26:
☺️ U have Realtek RTS5129 card reader. I too have that reader, not faced the problem mentioned by you but still @0xFireWolf can help you with that

mylife1234-cmd, 2021-09-26:
Thanks for your comment 
Thank you very much

gnpaone, 2021-09-26:
U can follow [InsanelyMac](https://www.insanelymac.com/forum/topic/348130-realtek-pcie-card-reader-driver-for-macos/page/7/) and report it there too alongwith logs (check [FAQ](https://github.com/0xFireWolf/RealtekCardReader/blob/main/Docs/FAQ.md))

mylife1234-cmd, 2021-09-26:
Thanks you so much

mylife1234-cmd, 2022-02-20:
Hay @0xFireWolf  I using Realtek Card reader0.8.6, I shut down my computer, but computer restart
Can't you fix it ?