0xFireWolf/RealtekCardReader #16: SD Card compatibility.

CLAY-BIOS, 2021-10-01:
@0xFireWolf First of all, thank you for your contribution!
1.SD Card compatibility.
My card reader is RTS5227,I tested two SD cards, one SanDisk and one banq. In sinetek-rtsx.kext, both cards can be recognized normally.
However, in realtekcardreader.kext, only SanDisk cards can be recognized, and both cards have error messages in the log.
[image]
[image]

SanDisk：
RTSX: IOReturn RealtekSDXCSlot::tuningRxCommand(UInt8) Error: Failed to send the CMD19 and get the response. Error = 0xe00002bc.
[rtsx-SanDisk.log](https://github.com/0xFireWolf/RealtekCardReader/files/7269332/rtsx-SanDisk.log)

BANQ：
RTSX: IOReturn RealtekSDXCSlot::tuningRxCommand(UInt8) Error: Failed to send the CMD19 and get the response. Error = 0xe00002bc.
RTSX: IOReturn RealtekSDXCSlot::tuningRx() Error: Rx Phase Map cannot be zero.
[rtsx-BANQ.log](https://github.com/0xFireWolf/RealtekCardReader/files/7269338/rtsx-BANQ.log)

2.Whether EMMC card can be supported.
I have an SD card using EMMC chip. It can work normally in windows and Linux. Using SDIO protocol, the read and write speed in win10 is about 40MB / s.
Because the driver is ported from Linux, you should be able to add support.
[image]
[rtsx-EMMC.log](https://github.com/0xFireWolf/RealtekCardReader/files/7269379/rtsx-EMMC.log)

0xFireWolf, 2021-10-01:
> SanDisk：
> RTSX: IOReturn RealtekSDXCSlot::tuningRxCommand(UInt8) Error: Failed to send the CMD19 and get the response. Error = 0xe00002bc.

Based on your kernel log, your SanDisk card was recognized properly. 
You don't need to worry about those errors related to CMD19, as long as the tuning process can finish successfully.

> BANQ：
> RTSX: IOReturn RealtekSDXCSlot::tuningRxCommand(UInt8) Error: Failed to send the CMD19 and get the response. Error = 0xe00002bc.
> RTSX: IOReturn RealtekSDXCSlot::tuningRx() Error: Rx Phase Map cannot be zero.

I was not able to download the second log `rtsx-BANQ.log`. Github returned back a 404 error.
However, the error message indicated that the tuning process has failed.
It might be related to that specific card. Could you please upload the kernel log again?
Also please try the BANQ card with the boot argument `-iosd3v3` and see if the card can be initialized properly.
This boot argument will initialize all cards at 3.3V, so UHS-I cards will run in either the default or the high speed mode.
Note that Sinetek's driver does not support UHS-I modes.

> Whether EMMC card can be supported.

Sorry, I am not interested in MMC/EMMC cards, so I won't allocate time for them.

CLAY-BIOS, 2021-10-01:
> Based on your kernel log, your SanDisk card was recognized properly. You don't need to worry about those errors related to CMD19, as long as the tuning process can finish successfully.
>
>
> I was not [...]

thank you！
`-iosd3v3`  valid.
Also: these logs may be useful to you.
[rtsx-BANQ.log](https://github.com/0xFireWolf/RealtekCardReader/files/7269986/rtsx-BANQ.log)
[rtsx-BANQ--iosd3v3.log](https://github.com/0xFireWolf/RealtekCardReader/files/7269988/rtsx-BANQ--iosd3v3.log)

0xFireWolf, 2021-12-27:
If you are still interested in testing the driver, please try the attached one with your BANQ card. 

The new fallback mechanism allows the driver to try to initialize your card at a lower speed mode after it has failed to initialize that card at the ultra high speed mode. Please check whether your BANQ card can be recognized properly **WITHOUT** the `-iosd3v3` boot argument and upload the kernel log as well as the IOReg dump for debugging purposes.

[RealtekCardReader_0.9.7_43b5887_DEBUG.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7779276/RealtekCardReader_0.9.7_43b5887_DEBUG.zip)

closed 2022-05-16 by 0xFireWolf