0xFireWolf/RealtekCardReader #18: RTS5227 Timeouts / Mount Errors

joshuaseltzer, 2021-10-06:
First off: thanks again for porting this over to macOS!

I got the latest nightly (https://github.com/0xFireWolf/RealtekCardReader/actions/runs/1218661751) installed on my machine:
- Lenovo T440s with RTS5227 built-in
- macOS Mojave 10.14.6
- Clover bootloader (using kext injection for all of my kexts, including yours)

I tried 5-6 SD cards I had lying around. 4 of them were 32gb cards and worked flawlessly. However, I noticed that a 128gb card I had did not mount correctly. I captured the logs from the moment I plugged it in, to the end of when it was printing errors (see attached). Line 11634 looks particularly interesting...
[RTS5227_Mount_Error.txt](https://github.com/0xFireWolf/RealtekCardReader/files/7293230/RTS5227_Mount_Error.txt)

I have a few other cards too I'd like to test with this machine.

gnpaone, 2021-10-07:
It initializes all cards at 3.3V, so cards can work at the default or the high speed mode but I think there are no downsides, probably transfer speeds will be reduced, not sure about it

joshuaseltzer, 2021-10-09:
> All UHS-I cards will be working at the default or the high speed mode with the boot argument `-iosd3v3`, so their speeds are limited to about 12.5MB/s or 25MB/s.
> 
> Your kernel log is incomplete. Could you please do a fresh reboot, insert the 128GB card that cannot be recognized, and dump the kernel log?

Sure. That was exactly what I had done the first time but maybe something got cut off. I’ll boot it up again tomorrow without that flag and capture the logs.

0xFireWolf, 2021-12-27:
If you are still interested in testing the driver, please try the attached one with your 128GB card. 

The new fallback mechanism allows the driver to try to initialize your card at a lower speed mode after it has failed to initialize that card at the ultra high speed mode. Please check whether your 128GB card can be recognized properly **WITHOUT** the `-iosd3v3` boot argument and upload the kernel log as well as the IOReg dump for debugging purposes.

[RealtekCardReader_0.9.7_43b5887_DEBUG.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7779276/RealtekCardReader_0.9.7_43b5887_DEBUG.zip)

The previous kernel logs you provided are incomplete. I have checked all error messages in your log, but I suspect that there should be several more errors in between them. I guess you are using macOS Catalina or earlier systems. If so, open the terminal before you insert the card, type the command `log stream | grep RTSX`, and then insert your 128GB card. Wait for about 30 seconds until the terminal does not print new lines. Click the `Shell` menu and save the terminal output as a file (`Export Text As...`).

joshuaseltzer, 2021-12-28:
@0xFireWolf Sure! I'm still willing to help test.

Here's the log output from terminal using that very same SD card with your latest build linked above. This time it mounted correctly! I also confirmed that I'm no longer using the `-iosd3v3` boot argument so I'm guessing it's using the correct speed as well.

Here's the log output:
[RTSX_128gb_2021-12-27.txt](https://github.com/0xFireWolf/RealtekCardReader/files/7782332/RTSX_128gb_2021-12-27.txt)

Let me know if you want me to test anything else! Thanks again!

0xFireWolf, 2021-12-28:
@joshuaseltzer Thanks for the quick feedback.

```
// Line 8107
RTSX: bool IOSDCard::initUltraHighSpeedMode() DInfo: Tuning has finished.
RTSX: bool IOSDCard::initUltraHighSpeedMode() DInfo: The card has been initialized with the ultra high speed mode enabled.
RTSX: bool IOSDHostDriver::attachCardAtFrequency(UInt32) DInfo: The card has been initialized successfully.
```

Yes, your log shows that the card has been initialized at the ultra high speed mode successfully.
I guess that's all for the test. I will close this issue and thank you again for joining the beta testing.

closed 2021-12-28 by 0xFireWolf

joshuaseltzer, 2021-12-28:
> @joshuaseltzer Thanks for the quick feedback.
>
> ```
> // Line 8107
> RTSX: bool IOSDCard::initUltraHighSpeedMode() DInfo: Tuning has finished.
> RTSX: bool IOSDCard::initUltraHighSpeedMode() DInfo: The [...]

Awesome, thank you. I will keep an eye out for the official release for 0.9.7.