0xFireWolf/RealtekCardReader #20: RTS5260 Does not seem to work

dursonATWIT, 2021-10-20:
The sd reader shows up in System Information, but  does not show any sd cards I tested (16-128GB Kingston/Microcenter branded) in System Information or Disk Utility.

[image]

[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/7383739/rtsx.log)
Thank you for showing some love to users with PCIE SD card readers!

0xFireWolf, 2021-12-28:
Your kernel log is incomplete. Please refer to the FAQ to make sure that you have the complete log.

dursonATWIT, 2022-01-04:
For some reason I'm not able to get a complete log on Monterey with debug enhancer and proper boot arguments. I'll upload what I got this time but given it's incomplete I'm not sure how much use it will be, I will try and get a complete log in Catalina soon
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/7805616/rtsx.log)

dursonATWIT, 2022-01-30:
I got a pretty long log in Catalina and using debug enhancer. I hope this log is helpful.
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/7965116/rtsx.log)