0xFireWolf/RealtekCardReader #22: After macos wakeup, the system usb setting  increase one sd card item

nabaonan, 2021-11-04:
- Sleep 
- Wake up
- See system setting  usb  there will be added a sd item which called CRW

nabaonan, 2021-11-05:
[image]

nabaonan, 2021-11-06:
[image]

nabaonan, 2021-11-06:
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/7490325/rtsx.log)

parnz, 2021-12-09:
I also have this problem and it doesn't detect sd card when insert at all.

[image]

dblhlx, 2021-12-10:
I have this problem too. The card is RTS5129.

0xFireWolf, 2021-12-11:
This is a known issue of RTS5129 (probably RTS5139 and RTS5179 as well).

My suspicion is that the USB card reader got disconnected when **some** laptops went to sleep, but the driver still assumes that the device remains connected thus holding a reference to it. When your laptop wakes up, the device gets reconnected, so the USB driver stack publishes a new instance of your device, but the old instance is still there because the driver holds a reference to it. Unfortunately, I do not have RTS5129 on my side, so I cannot test my hypothesis. There might be a quick yet dirty fix that disables the power management routines for RTS5129 if a special boot argument is provided, so the driver tears down when your laptop goes to sleep, and a new driver instance starts when your laptop wakes up.

dblhlx, 2021-12-11:
Thanks so much for you excellent work! I did not expect the card reader will be useful when setting the laptop as a Hackintosh. To my surprise, I can quickly transfer files using the driver you wrote. Your hypothesis is very likely true. My laptop is a Gateway GWTN156-3BK, not a common brand now. I disabled the dGPU (nvidia RTS 2060) using SSDT-dGPU-off.aml, but it always draws significant power after waking up. I need to use SSDT-PTSWAK.aml to turn the dGPU on when the laptop goes to sleep and turn it off when the computer wakes up.

If it is easy to implement your idea, I am more than happy to test your new code. Again thank you very much for your excellent work!

0xFireWolf, 2021-12-27:
@dblhlx Thanks for being interested in testing this driver. 

Please try the attached driver and add the boot argument `-rtsxnopm` to disable the power management routine. Please check if the issue persists. 

Specifically, ...
- Do you observe a kernel panic after the laptop wakes up?
- If you open the IORegistryExplorer before the laptop goes to sleep, can you observe a new USB card reader object that shows up in IORegistry after the laptop wakes up?
- Is your USB card reader still working after the laptop wakes up?

Please upload the kernel log dumped after the laptop wakes up as well.

[RealtekCardReader_0.9.7_0a0d816_DEBUG.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7781973/RealtekCardReader_0.9.7_0a0d816_DEBUG.zip)

dblhlx, 2021-12-29:
I tested the attached kext. Here are the results:
1. No kernel panic was observed after wake-up.
2. A new card reader object showed up in IORegistry after each wake-up (attached).
3. USB card reader still works after wake-up.

Thanks for your excellent work. Please let me know if more testing needed.

[image]
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/7786985/rtsx.log)

0xFireWolf, 2021-12-29:
@dblhlx Thanks for the feedback. 

What are the other three `USB2.0-CRW` objects in your screenshot? Did you wake up your laptop only once or for 4 times?

Do you still have the issue that a new `USB2.0-CRW` entry shows up under the USB Device Tree in System Information?
If so, could you please do the following steps and upload the IOReg dump?
- Reboot your computer. Do not insert any SD cards.
- Open the IORegistryExplorer and keep the window open.
- Put your laptop into sleep and wait for one or two minutes.
- Wake up your laptop.
- Save the IOReg dump and upload it.

Thank you.

dblhlx, 2021-12-29:
In my last post, I waked up my laptop for four times. I opened IORegistryExplorer after the third wake-up.

I followed your instruction this time. I rebooted my laptop and openedIORegistryExplorer, then I put the laptop into sleep. The laptop waked up by itself after a couple of minutes, and I save the IOReg (attached) after that. I don't understand what IOReg dump is. I just did File -> Save as. Please let me know if it is not what you need.

[IORegistry.ioreg.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7787170/IORegistry.ioreg.zip)

0xFireWolf, 2021-12-29:
@dblhlx Yes, that's the IOReg dump I need. 

With the boot argument `-rtsxnopm`, the card reader was disconnected and the driver was also terminated when your laptop went to sleep, because all entries under `HS02` were red in the IORegistry. So, it seems that your problem is resolved. Can you confirm that there aren't multiple `USB2.0-CRW` entries under `USB Device Tree` in System Information? Thank you.

dblhlx, 2021-12-29:
I still see multiple entries of USB2.0-CRW with the boot argument -rtsxnopm. I can also see multiple USB2.0-CRW@14200000/RealtekUSBCardReaderController entries but a single RealtekUSBSDXCSlot entry in the IORegistryExplorer screenshot of my previous post. Nonetheless, I consider this is a cosmetic issue because the card reader is functional.

Thank you very much!
[image]

mac-os-x, 2022-01-24:
Hello, I was able to identify the sd card with the same kext, but when I say restart after sleep, the system shuts down too late. I have tested it many times and such a problem occurs in both Big Sur and Monterey. @0xFireWolf

mac-os-x, 2022-01-24:
[image]

prud281, 2022-03-28:
-rtsxnopm argument doesn't hel
[image]
p

maverx72, 2022-04-10:
> -rtsxnopm argument doesn't hel
> [image]
> p

Same for me

maverx72, 2022-04-10:
With the boot argument -rtsxnopm there is no change

maverx72, 2022-04-10:
bluetooth still can't work around 15 minutes, Card reader work, Card reader USB2.0-CRW device still increase and shutdown still slowed after sleep awake operation.

toxa70x4, 2022-05-03:
Very relevant, waiting for a fix and a new version.
Many thanks to the author.

franksfc, 2022-05-31:
Is there any chance to fix it? I can offer any information concerned.

lslqtz, 2022-07-12:
Same issue, I thought it was caused by the system version,so I tried these versions but still had problems. (macOS 11.6.7/macOS 12.0 beta 5/macOS 12.0 beta 6/macOS 12.1/macOS 12.4)

bismillah-100, 2025-10-24:
I've tried to fix it and the issue has been fixed for me.

[RealtekCardReader_0.9.7__RELEASE-FIX.zip](https://github.com/user-attachments/files/23113730/RealtekCardReader_0.9.7__RELEASE-FIX.zip)

0xFireWolf, 2025-10-24:
> > for most cases, its should be fine like the original one.
> > I’ll make a pull request if someone is still here.

@bismillah-100 Thanks for taking the time to investigate the issue. However, I don't have any USB card readers, so I could not test it myself. 

Since you suggested that some objects were not released properly (which is a memory leak), It would be great if you make a pull request anyway, and I will merge it and passively wait for other folks to report problems.