0xFireWolf/RealtekCardReader #24: Not recognizable as a built-in card reader device

git897, 2021-11-29:
First thank you for your help that i can use the sd card.  But the computer not recognizable as a built-in card reader device.It just display in storage page.i also use the cardreader friend kext.but it doesn't work
[image]
[image]
[image]

0xFireWolf, 2021-11-30:
Please identify your SD card reader.
i.e. Device name, vendor ID, device ID, etc.

git897, 2021-12-01:
[image]

0xFireWolf, 2021-12-01:
Please refer to the [README](https://github.com/0xFireWolf/RealtekCardReader#features) and make sure that you have both RealtekCardReader and RealtekCardReaderFriend installed on your computer.

closed 2021-12-01 by 0xFireWolf