0xFireWolf/RealtekCardReader #25: 9.7

jmacie, 2021-11-30:
Hi.Firewolf, I had asked, on insanelymac forum, if you could compile 9.7 without the  debug part. I am able to use this kext, but it slows my boot time to a slog. I would try to compile it if I could , but I'm no coder. Thanks for the kext, It makes my  little hack that much better.

0xFireWolf, 2021-11-30:
Sorry for the late reply. I was busy recently and didn’t check the posts on InsanelyMac. 

Nightly builds contain both the DEBUG and the RELEASE version of the driver. You can download them from the Artifacts section in each workflow run. For example, the latest workflow run is https://github.com/0xFireWolf/RealtekCardReader/actions/runs/1218661751.

closed 2021-11-30 by 0xFireWolf

jmacie, 2021-11-30:
So awesome! Thanks for all you do! Take care, Happy Holidays!