0xFireWolf/RealtekCardReader #26: RTS5287 support

Giovix92, 2021-12-01:
Hello there, I have an RTS5287 (aka RTL8411B) card reader, and I can't get it to work at all. Reading the known issues page, I've seen that the device is almost "killed" since the start. Is there anything I could do/provide to help enabling it? An IOReg maybe?

Thank you!
- Giovix92

0xFireWolf, 2021-12-27:
Please provide the kernel log as per [FAQs](https://github.com/0xFireWolf/RealtekCardReader/blob/main/Docs/FAQ.md#how-do-i-dump-the-log-produced-by-the-driver).

Giovix92, 2022-01-02:
Here you are:

`❯ kextstat | grep rtsx
   86    0 0                  0x106000   0x106000   science.firewolf.rtsx (0.9.6) 896FEAB9-732E-3779-B18D-F714BF427168 <33 24 16 6 3>`

`❯ sudo dmesg | grep "rtsx"
[  110.846934]: successfully parsed 1/4 entries for file rtsx.logIOConsoleUsers: time(0) 0->0, lin 1, llk 0,`

Those are the only things I could achieve right now. The "kernel log" with the command `sudo dmesg | grep RTSX > ~/Desktop/rtsx.log` just returns an empty log file.
Used the 0.9.6 DEBUG version + these boot arguments: `-wegnoegpu igfxonln=1 brcmfx-country=#a alcid=13 msgbuf=10485760`.

Giovix92, 2022-01-13:
Tried the latest artifact, with RealtekCardReaderFriend the PCIe card reader is detected under System Informations / Card Readers, though still no useful dmesg available.
Anything else I can do?

seffs, 2022-01-19:
@Giovix92 you forgot to use the DEBUG version of the kext

@0xFireWolf Attached is the dump of my machine running Mojave 10.14.6. Steps to reproduce:

1. Boot up/log in w/o SD card
2. Insert SD card
3. Eject SD card

As a quick note: this reader gets recognized with the classic Sinetek-rtsx kext, if that helps you in any form. 

If you need any more info let me know
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/7900240/rtsx.log)

Giovix92, 2022-01-19:
> @Giovix92 you forgot to use the DEBUG version of the kext
> 
> @0xFireWolf Attached is the dump of my machine running Mojave 10.14.6. Steps to reproduce:
> 
> 1. Boot up/log in w/o SD card
> 2. Insert SD card
> 3. Eject SD card
> 
> Output from sysreport:
> 
> ```
> Built in SD Card Reader:
> 
>   Vendor ID:	0x10ec
>   Device ID:	0x5287
>   Subsystem Vendor ID:	0x1025
>   Subsystem ID:	0x1094
>   Revision:	0x0001
>   Link Width:	x1
>   Link Speed:	2.5 GT/s
> ```
> 
> As a quick note: this reader is "functional" with the classic Sinetek-rtsx kext, if that helps you in any form.
> 
> If you need any more info let me know [rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/7900240/rtsx.log)

As I've stated in a previous reply, I actually did use the DEBUG version, since it was pointed out on the FAQs, plus I did put the `msgbuf` boot argument.
I'm running Monterey 12.1, by the way.

I'm getting the same sysreport result, just no dmesg log.

seffs, 2022-01-19:
> I'm getting the same sysreport result, just no dmesg log.

I have these sysargs in my boot parameters (OpenCore). Maybe "debug" will do it for you?

`keepsyms=1 debug=0x100 swd_panic=1 `

WalaceP, 2023-04-07:
I'm a beginner. can check?
[image]