0xFireWolf/RealtekCardReader #28: Monterey supported?

BlvckBytes, 2021-12-11:
Hey there!

Will this driver work with macOS Monterey? I can't seem to get it to work, just wondering whether it's my mistake, or if the kext is not yet supporting this version of the OS.

0xFireWolf, 2021-12-11:
Yes, the driver is working properly on my machine with macOS Monterey 12.0.1 installed.

closed 2021-12-11 by 0xFireWolf

BlvckBytes, 2021-12-11:
Seems weird then, doesn't work at all on my machine. I'm using a Realtek RTS5260, and both `RealtekCardReader.kext` and `RealtekCardReaderFriend.kext` are loaded (verified with kextstat). I'm on an XPS 9700 4K. Any ideas what I'm missing
[EFI.zip](https://github.com/0xFireWolf/RealtekCardReader/files/7697495/EFI.zip)
?

0xFireWolf, 2021-12-11:
Based on the feedback from other users, the driver may fail to initialize SDXC cards on some RTS5260 models.
It may help if you add the boot argument `-iosd3v3` to initialize all cards at 3.3V.

BlvckBytes, 2021-12-11:
Still not working, I think I'm just out of luck with this... :/.