0xFireWolf/RealtekCardReader #30: Doesn't work on Latitude 7370

mx4address, 2022-02-17:
I tried to use it on my Latitude 7370 running Catalina, with both kexts injected, it recognizes the SD Card reader, but no SD Card shows up on Disk Utility or anything.
If someone can help.. 

Built in SD Card Reader:

  Vendor ID:	0x10ec
  Device ID:	0x525a
  Subsystem Vendor ID:	0x1028
  Subsystem ID:	0x071b
  Revision:	0x0001
  Link Width:	x1
  Link Speed:	5.0 GT/s

mx4address, 2022-02-17:
RTS525A is my card reader

0xFireWolf, 2022-05-16:
Please attach the **complete** kernel log as per [FAQ](https://github.com/0xFireWolf/RealtekCardReader/blob/main/Docs/FAQ.md#how-do-i-dump-the-log-produced-by-the-driver). Issues without kernel logs or with incomplete kernel logs will be ignored and closed.

closed 2022-09-12 by 0xFireWolf