0xFireWolf/RealtekCardReader #33: RTS5227 PCI Express Card Reader

chris1111, 2022-05-14:
Not working here on my HP Probook 650 G1
It work with Sinetek-rtsx.kext but I would have liked to use your kext
This is the Infos from Hackintool

`04:00.0 10EC 5227 103C 2101 L0s+L1 Realtek Semiconductor Co., Ltd RTS5227 PCI Express Card Reader                    Unassigned class     ???                  /pci10ec,5227@0 pci10ec,5227    PciRoot(0x0)/Pci(0x1C,0x7)/Pci(0x0,0x0) `

0xFireWolf, 2022-05-16:
Please attach the **complete** kernel log as per [FAQ](https://github.com/0xFireWolf/RealtekCardReader/blob/main/Docs/FAQ.md#how-do-i-dump-the-log-produced-by-the-driver). Issues without kernel logs or with incomplete kernel logs will be ignored and closed.

chris1111, 2022-05-16:
Ok sorry for the Faq and the Log but now with the boot args  `msgbuf=10485760`
the system not boot any more, kenel panic ?

chris1111, 2022-05-16:
Sorry My fault issue booting is resolved, here Log 

[rtsx.log.zip](https://github.com/0xFireWolf/RealtekCardReader/files/8703751/rtsx.log.zip)

chris1111, 2022-05-16:
strangely since I managed to boot with the debug the SD card mounts well and moreover I can eject and put back a different one which I cannot do with the kext sinetek.

Now it seems to work fine only it doesn't show up in About This Mac, I have to enter DeviceProperties

[image]

the release kext is loaded correctly
[image]

0xFireWolf, 2022-05-16:
Your attached kernel log has no error messages and all recorded I/O transactions have succeeded.

> Now it seems to work fine only it doesn't show up in About This Mac, I have to enter DeviceProperties

Device Properties are not needed.

As mentioned in the [Features](https://github.com/0xFireWolf/RealtekCardReader#features) section in the README file, you need to install the companion Lilu plugin [RealtekCardReaderFriend](https://github.com/0xFireWolf/RealtekCardReaderFriend/) to make your card reader recognized as a built-in one.

chris1111, 2022-05-16:
ok Now I have an update monterey and I will test later tonight

closed 2022-05-16 by chris1111

chris1111, 2022-05-17:
Yes it work Thank you very mutch
[image]

Agustine808, 2023-07-31:
[image]
how to remove type which is mentioned here as "unsigned class"?

Agustine808, 2023-07-31:
> strangely since I managed to boot with the debug the SD card mounts well and moreover I can eject and put back a different one which I cannot do with the kext sinetek.
>
> Now it seems to work fine [...]

how did you managed to hands on memory card type "secure digital memory card? when mine is unassigned class in type!