0xFireWolf/RealtekCardReader #37: Could not format SD card with APFS (RTS5527)

jogerj, 2022-08-21:
I have a Samsung EVO Select 256GB. I tried to format with disk utility

```
Erasing “Samsung GE4S5 Media” (disk4) and creating “macSD”

Unmounting disk
Creating the partition map
Waiting for partitions to activate
Formatting disk4s2 as Mac OS Extended (Journaled) with name macSD
Initialized /dev/rdisk4s2 as a 238 GB case-insensitive HFS Plus volume with a 24576k journal

Mounting disk
Creating a new empty APFS Container
Unmounting Volumes
Switching disk4s2 to APFS
Creating APFS Container
tx_flush:1063: rdisk4s2 xid 1 sync/barrier failed: 45
tx_mgr_free_tx:208: rdisk4s2 Trash unfinished pending tx, xid range = 0x1 - 0x1
newfs_apfs: unable to format /dev/disk4s2: Operation not supported
failed to finish first transaction: 45 - Operation not supported

An internal error has occurred. : (-69626)

Operation failed…
```
tried `new_apfs /dev/disk4`
```
failed to finish first transaction: 45 - Operation not supported
tx_flush:1063: rdisk4 xid 1 sync/barrier failed: 45
tx_mgr_free_tx:208: rdisk4 Trash unfinished pending tx, xid range = 0x1 - 0x1
newfs_apfs: unable to format /dev/disk4: Operation not supported
```

dmesg | grep -i card
```
05:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader (rev 01)
```

Allegedly it's something that should be possible according to this video https://www.youtube.com/watch?v=Adpoy75iJoI

I unfortunately don't have a mac so I can't test it, the closest thing I did was use https://github.com/linux-apfs/apfsprogs 
The resulting sdcard was read only, even though mac recognizes it, it won't write to it. I really can't tell if it's an issue with the driver, or something else.

Formatting with HFS+ works flawlessly.

0xFireWolf, 2022-08-21:
I was able to reproduce this issue on my machine (RTS525A) with a SanDisk SDXC card. 

I also checked the kernel log, but there was no error message on the driver side when the APFS subsystem failed to create the filesystem on the SD card. Formatting the card with ExFAT or HFS+ works perfectly fine. As such, I will not investigate this issue further.

closed 2022-08-21 by 0xFireWolf

ffffwh, 2022-10-31:
I've formatted a micro SD card with a generic card reader as 'APFS Encrypted'. It cannot be written with the internal reader of NUC 8.

```
Finder:
The operation can’t be completed.
An unexpected error occurred (error code -1426).
```

```
kernel (apfs)
tx_flush_thread:1582: disk3 tx_flush(29) failed, 45, # 317
```

Some workarounds could be
- use a encrypted disk image on HFS+
- create HFS+ encrypted with an older OS