0xFireWolf/RealtekCardReader #38: Not work on my latitude 5175 RTS525A

Einstonoy, 2022-08-21:
I tried to use it on my Latitude 5175 running MacOS 12 with both kexts injected, it recognizes the SD Card reader, but no SD Card shows up on Disk Utility or anything.
If someone can help..

The installed test shows the drive seems to  been installed.
but when I insert tf card ,it do not response at all.

the kernel log as follow
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/9389693/rtsx.log)
Really appreciated if someone could help me. 
Thx a lot!

0xFireWolf, 2022-08-21:
Please identify your card. e.g., Brand, Type (SDSC/HC/XC?), Speed Class.

Your kernel log shows that your card does not respond to CMD8 and the initial ACMD41 (which triggers the card initialization sequence). Try to add the boot argument `rtsxdssc=30` to see whether the card initialization sequence still fails.

Einstonoy, 2022-08-21:
Appreciated a lot for your soon respond! info as follow,
Sandisk sdxc speed class  least 10, whose label has v30

I wil test the argument you recommended, may reply in minutes. Thx a lot again!

Einstonoy, 2022-08-21:
Dear Wolf,
The new log implies the initialization remains failed after I added the argument you suggested. Wonder is there any ways I can try to make it work. 
Best wishes! Thanks a lot!

The new log as follow,
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/9389767/rtsx.log)

Einstonoy, 2022-08-21:
Maybe I should  test a lower speed grade tf card?

0xFireWolf, 2022-08-21:
> The new log implies the initialization remains failed after I added the argument you suggested.

Indeed, your card still does not respond to CMD8 and ACMD41.

> Maybe I should test a lower speed grade tf card?

If possible, please do so, because I am not sure whether this is a card-specific issue or a bug in the driver at this moment.

Einstonoy, 2022-08-21:
I tried just on my sdhc class4 grade, still failed. Thought it could not be a card-specific issue.  Trying another middle grade card now, upgrading in minutes.

Appreciated for you help!

Einstonoy, 2022-08-21:
Intersetingly, according drive in the system info in PCIE and SD reader panels is missing after I insert the class4 tf before power on. Which is normally informed before.

0xFireWolf, 2022-08-21:
Please insert those SD cards after you can see your desktop.

Einstonoy, 2022-08-21:
OK>>>>....

It seems all grade kinds tf are not work after the third test, the result may mean that the drive could be incompatible with my laptop.

Einstonoy, 2022-08-21:
> Please insert those SD cards after you can see your desktop.
Got it, I try it again now

Einstonoy, 2022-08-21:
Yeah>>>.... 

After boot insert is not work either, and drive in the system info is gone after insert.

Einstonoy, 2022-08-21:
Dear Wolf

The latest log is followed, it would be so grateful if you got any adivce to me.
[rtsx.log](https://github.com/0xFireWolf/RealtekCardReader/files/9389815/rtsx.log)

Einstonoy, 2022-08-22:
Dear Wolf
Sincerely thanks for you help,  looking forward to your updating news or new version of this sd card drive.
By Einstonoy

closed 2022-08-24 by 0xFireWolf