0xFireWolf/RealtekCardReader #40: Kext wont load for Mojave

mslifcak, 2022-09-15:
RealtekCardReader.kext does not load on 10.14.6 Mojave
Observed using Clover 5148 or OpenCore 0.8.3. 
  There is no such problem for 10.15, 11, 12 or 13.

Hardware:  Dell Latitude 7480, Board 00F6D3
Realtek 525A Vendor ID: 0x10ec   Device ID: 0x525a
Subsystem Vendor ID: 0x1028  Subsystem ID: 0x07a0  Revision ID: 0x0001

Clover misc/preboot.log shows no error for the kext

OpenCore efi log for Mojave has one line:
  OC: Prelinked Injection (RealtekCardReader.kext (Realtek SDCard Reader (0xFireWolf)) - Invalid Parameter

DebugEnhancer and msgbuf boot arg tried for both Clover and OpenCore.  No RTSX lines in kernel log

gujiangjiang, 2022-10-29:
Dont foget RealtekCardReaderFriend.kext

mslifcak, 2022-11-04:
Problem solved with this combination:
 RealtekCardReader.kext 0.9.7
 RealtekCardReaderFriend.kext 1.0.4
 
 Many thanks to all who helped build the working releases.

closed 2022-11-04 by mslifcak