0xFireWolf/RealtekCardReader #44: RTS5227

Sniki, 2022-12-17:
Hi @0xFireWolf.

I just checked the readme and i do have a RTS5227 card reader on my T440/T440S/X240/T440P laptops that i have.

Right now i am running latest release of Big Sur on my T440P;
The system would randomly kernel panic/instant reboot and i noticed once it boots up again, it shows on the apple problem report that it’s this kext, i noticed that logs are incomplete on the known issues section.

If you don’t mind just giving me quick steps on how i would normally collect the logs for your case, i would be able to test and provide logs to fix the issues.

Thanks in advance.

joshuaseltzer, 2023-01-16:
@Sniki I'm running Mojave with my T440s (which uses the same RTS5227 card) alongside the latest release of this kext (0.9.7). I have not had any KPs with this kext installed. See issue #18 where @0xFireWolf helped me debug and add better support for this card, which might also help you generate logs for him to further investigate.

I have my kext installed via injection using Clover bootloader, and I also have [RealtekCardReaderFriend](https://github.com/0xFireWolf/RealtekCardReaderFriend) installed although I do not think it's necessary.

Sniki, 2023-01-16:
@joshuaseltzer then it seems that the issue is specific to newer macOS versions, it happens to me on Big Sur and Monterey.
Same scenario happened on my X240/T440P/T440S from what I've tested so far, i also add RealtekCardReaderFriend.kext as it makes your card reader show on System Report > Card Reader.
Apart from this random intermitent Kernel Panic into an instant reboot, the kext seems to be working perfectly fine.

Thanks for the support.

joshuaseltzer, 2023-01-17:
> @joshuaseltzer then it seems that the issue is specific to newer macOS versions, it happens to me on Big Sur and Monterey. Same scenario happened on my X240/T440P/T440S from what I've tested so far, [...]

Then maybe it is related to the specific macOS version, or maybe the bootloader (yes, I'm still on Clover).

Also, I should mention, I rarely use the SD Card reader on my T440s. I've probably only used it 2 or 3 times since the kext was released. So if your issue is only when a card is inserted, maybe mine would have the same issue. My system has had almost 8 months of uptime at this point though, so it's pretty rock solid stable.