0xFireWolf/RealtekCardReader #48: Venture working 0.9.6

N3kowarriorCZenchilada, 2023-10-30:
Hi I have the 0.9.6 with friend kext working on mac os Venture. If i update to 0.9.7 kext stops working please mark this (0.9.6) version with this card reader as compatible with mac os Venture.

I copied some data from sd card but i cant demostrate it because it was school work. And I don't have any sd card at home.

[image]

N3kowarriorCZenchilada, 2023-10-30:
And i have it in device properties as this:
[image]

0xFireWolf, 2023-11-05:
Cannot reproduce the issue with 0.9.7 on macOS Ventura 13.6. If you believe that this is a bug in 0.9.7, please refer to the FAQ and upload all necessary logs.

Side Note: You can remove those device properties. They are not needed by this driver at all.

closed 2023-11-05 by 0xFireWolf