0xFireWolf/RealtekCardReader #55: Sleep transition timed out after 180 seconds while calling power state change callbacks

lxopencv, 2024-04-29:
OC0.9.9 0x10ec0x522a
```
panic(cpu 0 caller 0xffffff800199422e): Sleep transition timed out after 180 seconds while calling power state change callbacks. Suspected bundle: science.firewolf.rtsx. Thread 0x327f0.
Failure code:: 0x00000100 00000014
```

zearp, 2024-06-09:
First off, many thanks for this driver!  Been using this driver for years with different macOS versions with zero panics and reliable sdcard usage on my 8th gen Intel NUC's.

I also had a panic relating to sleep. I haven't found a way to reproduce it, will try more things if it happens again but I couldn't trigger it using a dozen or so sleep/wake cycles.

There was no card present and I have only ever had one panic, the one that just happened. Only chiming in to share my panic log in case it helps.

```
panic(cpu 0 caller 0xffffff8010d944ae): Sleep transition timed out after 180 seconds while calling power state change callbacks. Suspected bundle: science.firewolf.rtsx. Thread 0x2eb75.
Failure code:: 0x00000200 00000014

Backtracing specified thread
Panicked task 0xffffff953abb35d8: 198 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff9a062bd598, Frame : Return Address
0xffffffaa6f5ef790 : 0xffffff8010775657 mach_kernel : _machine_switch_context + 0xd7
0xffffffaa6ff53b30 : 0xffffff80106547f6 mach_kernel : _thread_block_reason + 0x1ff6
0xffffffaa6ff53bc0 : 0xffffff80106528f0 mach_kernel : _thread_block_reason + 0xf0
0xffffffaa6ff53c10 : 0xffffff801063f891 mach_kernel : _lck_mtx_sleep + 0x61
0xffffffaa6ff53c40 : 0xffffff8010d0ed90 mach_kernel : __ZN10IOWorkLoop9sleepGateEPvj + 0x80
0xffffffaa6ff53c80 : 0xffffff8010d1037e mach_kernel : __ZN13IOEventSource9sleepGateEPvj + 0x4e
0xffffffaa6ff53cb0 : 0xffffff80141ffdf1 science.firewolf.rtsx : __ZZ22IOCommandGateRunActionIZN27RealtekCardReaderController19onSDCardRemovedSyncE10BitOptionsIjEEUlvE_EiP13IOCommandGateT_ENUlP8OSObjectPvS9_S9_S9_E_8__invokeES8_S9_S9_S9_S9_ + 0x5d
0xffffffaa6ff53cf0 : 0xffffff8010d11cd8 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xa8
0xffffffaa6ff53d50 : 0xffffff80141ff64c science.firewolf.rtsx : __ZN27RealtekCardReaderController14prepareToSleepEv + 0x52
0xffffffaa6ff53d80 : 0xffffff80141f9cfb science.firewolf.rtsx : __ZN30RealtekPCICardReaderController14prepareToSleepEv + 0x1d
0xffffffaa6ff53dd0 : 0xffffff80141ff7a7 science.firewolf.rtsx : __ZN27RealtekCardReaderController13setPowerStateEmP9IOService + 0x45
0xffffffaa6ff53e00 : 0xffffff8010cf3a79 mach_kernel : __ZN9IOService19driverSetPowerStateEv + 0x1d9
0xffffffaa6ff53e70 : 0xffffff8010cf37dc mach_kernel : __ZN9IOService15pmDriverCalloutEPS_Pv + 0x9c
0xffffffaa6ff53ea0 : 0xffffff8010681bc8 mach_kernel : _thread_call_delayed_timer + 0x508
0xffffffaa6ff53ee0 : 0xffffff8010682c58 mach_kernel : _thread_call_delayed_timer + 0x1598
0xffffffaa6ff53fa0 : 0xffffff80105bf19e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         science.firewolf.rtsx(0.9.7)[ED226205-511C-3EE6-B8D5-31AB4B670EA1]@0xffffff80141f2000->0xffffff801428afff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[43511668-C040-3FBA-8CD6-25AD6E00FD90]@0xffffff8012968000->0xffffff8012999fff
            dependency: com.apple.iokit.IOStorageFamily(2.1)[1EEEAAAA-95C2-3606-8488-591BE79DB5B4]@0xffffff8012a9e000->0xffffff8012ab4fff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[E4CB5EAB-B685-3E1B-875A-CCC1CC176D34]@0xffffff8012d21000->0xffffff8012d9afff

Process name corresponding to current thread (0xffffff9a062bd598): kernel_task
Boot args: -v keepsyms=1 debug=0x100

Mac OS version:
23F79

Kernel version:
Darwin Kernel Version 23.5.0: Wed May  1 20:09:52 PDT 2024; root:xnu-10063.121.3~5/RELEASE_X86_64
Kernel UUID: 7B071FD2-9FDC-344E-8483-47EC972A105E
roots installed: 0
KernelCache slide: 0x0000000010200000
KernelCache base:  0xffffff8010400000
Kernel slide:      0x00000000102e4000
Kernel text base:  0xffffff80104e4000
__HIB  text base: 0xffffff8010300000
System model name: Macmini8,1 (Mac-7BA5B2DFE22DDD8C)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 22827331719834
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000014c2e6ff967a
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x000000149a90f182 0x0000000000000000
Compressor Info: 1% of compressed pages limit (OK) and 2% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff806d43f000 - 0xffffffa06d43f000
  . PGZ   : 0xffffff806d43f000 - 0xffffff806f440000
  . VM    : 0xffffff806f440000 - 0xffffff853bc3f000
  . RO    : 0xffffff853bc3f000 - 0xffffff86d543f000
  . GEN0  : 0xffffff86d543f000 - 0xffffff8ba1c3f000
  . GEN1  : 0xffffff8ba1c3f000 - 0xffffff906e43f000
  . GEN2  : 0xffffff906e43f000 - 0xffffff953ac3f000
  . GEN3  : 0xffffff953ac3f000 - 0xffffff9a0743f000
  . DATA  : 0xffffff9a0743f000 - 0xffffffa06d43f000
  Metadata: 0xffffffd032cd9000 - 0xffffffd052cd9000
  Bitmaps : 0xffffffd052cd9000 - 0xffffffd055cd9000
  Extra   : 0 - 0
```

This is on Sonoma 14.5, if there's anything I can do let me know. I might swap out the kexts for the debug version if that improves the panic logs in case it happens again.

mister-seffers, 2024-06-10:
Also having this issue on two systems (a Precision 7520 (Skylake) and a Precision 7530 (Coffee Lake)). Both run Sonoma 14.5 and am unable to either manually reproduce it or predict when it'll happen on either. 

Sleep/wake works 100% of the time in Windows and Linux, which rules out a hardware issue. If I disable RealtekCardReader and RealtekCardReaderFriend, macOS also sleeps/wakes with 100% reliability.

However, if I enable those kexts, both systems will sometimes sleep and wake just fine for multiple days in a row, then suddenly get this kernel panic during a sleep/wake cycle after many such successful cycles; or, it can happen the first time the systems try to go to sleep after booting. It is just as random and unpredictable on both. No variables I've tried (boot with card in, boot with card out, boot with card in and remove it, boot with card out and insert one, warm boot, cold boot, boot to windows/restart/boot macOS, etc.) have any impact on this behavior on either system.

The card reader _works_ in macOS on both systems, and shows up in System Information as expected. It just cannot reliably transition power states from wake to sleep or vice versa.

Given that the kext hasn't been updated in nearly 2 years, I don't have high hopes for this getting resolved so, unfortunately, I'm going to have to discontinue use; the pro of having a working card reader (and being able to say both systems are completely macOS compatible, as even Thunderbolt hotplug works on both) is not worth the con of having completely random panics I can neither predict nor mitigate.

beren12, 2025-06-16:
same with me on Sequoia. power management issue maybe?

zearp, 2026-11-15:
The panic has been fixed and also fixed the CPU not being able to enter C6 when no card is inserted, still have to fix C6 when a card is inserted but at least the panic is fixed and C6 is reached with nothing inserted. Had a little help from Claude as this is not my cup of tea. But after many hours of testing later I can it is fixed on my Intel NUCs (RTS522A) and can no longer reproduce the panic. If anyone is interested in compiling/testing their own and/or improving please let me know in the comments on the PR I made here: https://github.com/zearp/RealtekCardReader/commit/7ff367d46af500eaedafc08760c29ed820de2f6c